bigtreetech/Eddy #91: Eddy CAN instructions

serbiansky, 2024-11-06:
Since you have announced and start selling CAN version of Eddy can you please update the documentation for the CAN version? I am currently building a Formbot 2.4 kit but I would like to replace TAP with Eddy CAN to simplify the wiring and I already ordered the CAN version.

krautech, 2024-11-06:
I don't think the CAN version is shipping yet is it?

serbiansky, 2024-11-06:
I ordered on 1 November. Should receive in few days.

Erehr, 2024-11-06:
It will probably officialy debut during formnext (19-22.11) as one of the promo image show eddy silhouette.

Nomanpour, 2024-11-09:
I have received the BTT Eddy Duo,
I connected it to BTT EBB SB2209, and now I can see the uuid.
`Found canbus_uuid=91464af9d0e2, Application: CanBoot`

now I want to compile the klipper firmware to flash to the eddy.

[image]

What should I enter for CAN RX/TX?

BTTJUSTIN, 2024-12-24:
The documentation has been updated. Thanks again for pointing out the issue.

closed 2024-12-24 by BTTJUSTIN