bigtreetech/Eddy #92: BTT Eddy Duo CAN RX/TX GPIO Pins

Nomanpour, 2024-11-09:
Hello,

I connected it to BTT EBB SB2209, and now I can see the uuid.

Found canbus_uuid=91464af9d0e2, Application: CanBoot

now I want to compile the klipper firmware to flash to the eddy.

[image]

Which pins do I need to enter in `make menuconfig`?

Nomanpour, 2024-11-10:
I used the default 4 and 5 and it works fine!
the configuration just like the screenshot.

hergla, 2024-11-13:
Documentation needs to be updated fir Can usage.

SuperShipley, 2024-11-13:
Where on the SB2209 did you plug it in? I got my Eddy Duo yesterday and wasn't entirely sure which port would work.

Nomanpour, 2024-11-13:
[image]

> Where on the SB2209 did you plug it in? I got my Eddy Duo yesterday and wasn't entirely sure which port would work.

SuperShipley, 2024-11-14:
So I plugged it in there, have tried several times to get it into DFU mode with the boot button but the only way it shows up is if I plug it in via the USB cable not into the SB0000 as you mentioned. 

> ![image](https://private-user-images.githubusercontent.com/11517299/385974210-7c595379-89e0-4265-82d7-388794e5d55c.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3 [...]

Nomanpour, 2024-11-14:
You dont need the DFU mode,
just set the switch in CAN position, and connect the cable.
presume that you already have a working CANBUS.
after query the CAN, it shows you the uuid and application canboot so you can flash klipper by CAN

serbiansky, 2024-12-14:
@Nomanpour can you share your config file, I manage to flash katapult and klipper but when add below to config.cfg facing "mcu 'mcu': Unable to connect"

[mcu eddy]
canbus_uuid: 3cee77cba821
canbus_interface: can0

[temperature_sensor btt_eddy_mcu]
sensor_type: temperature_mcu 
sensor_mcu: eddy
min_temp: 10 
max_temp: 100

[probe_eddy_current btt_eddy]
sensor_type: ldc1612
z_offset: 2.5
i2c_mcu: eddy  
i2c_bus: i2c0f 

x_offset: 0
y_offset: 21.42

[temperature_probe btt_eddy]
sensor_type: Generic 3950
sensor_pin: eddy:gpio26
horizontal_move_z: 2

[bed_mesh]
horizontal_move_z: 2
speed: 200

mesh_min: 10, 10  
mesh_max: 220, 220 
probe_count: 9, 9
algorithm: bicubic

[safe_z_home]
home_xy_position: 175, 175 
z_hop: 10
z_hop_speed: 25
speed: 200

[force_move]
enable_force_move: True

Nomanpour, 2024-12-15:
`
[mcu eddy]
canbus_uuid=91464af9d0e2

[temperature_sensor btt_eddy_mcu]
sensor_type: temperature_mcu
sensor_mcu: eddy
min_temp: 10
max_temp: 100

[probe_eddy_current btt_eddy]
sensor_type: ldc1612
#z_offset: 2.5
#i2c_address:
i2c_mcu: eddy
i2c_bus: i2c0f

x_offset: 0
y_offset: 21.42

[temperature_probe btt_eddy]
sensor_type: Generic 3950
sensor_pin: eddy:gpio26
horizontal_move_z: 2
`

serbiansky, 2024-12-25:
Thank you for sharing!

Nomanpour, 2025-05-05:
For mine, Canboot was already installed, so I just flashed the Klipper.
Shouldn't be any problem if you install the Katapult correctly yourself.

Nomanpour, 2025-08-13:
Sorry, I don't have EBB36.
Maybe an issue with your termination resistor placement.

closed 2025-08-13 by Nomanpour

Nomanpour, 2025-08-13:
Yes, each device on the CAN Bus has its own unique ID.