bigtreetech/Eddy #96: Unable to flash Eddy DUo

Quinemc, 2024-11-22:
Hi , I'm having error to flash Eddy via DFU

Flashing out/klipper.uf2 to 2e8a:0003
sudo lib/rp2040_flash/rp2040_flash out/klipper.uf2

Loaded UF2 image with 149 pages
Found rp2040 device on USB bus 1 address 5
Flashing...
Resetting interface
Locking
Exiting XIP mode
Erasing
Flashing
Rebooting device
Command reboot failed with status 1: unknown command

Failed to flash to 2e8a:0003: Error running rp2040_flash

If the device is already in bootloader mode it can be flashed with the
following command:
  make flash FLASH_DEVICE=2e8a:0003

Alternatively, one can flash rp2040 boards like the Pico by manually
entering bootloader mode(hold bootsel button during powerup), mount the
device as a usb drive, and copy klipper.uf2 to the device.

make: *** [src/rp2040/Makefile:78: flash] Error 255

Nevio024, 2024-11-23:
I'm having exactly the same issue. I'm running an Eddy USB

Nevio024, 2024-11-25:
Hey @Quinemc , I just had the same issue and seem to have fixed it. 

Hold down the boot button on your eddy while connecting it to your computer. It should show up as "RP2040" or something similar in your file manager. 

You can then simply drag over "klipper.uf2". THe file "klipper.uf2" can be found in the folder ~/klipper/out/ on your raspberry pi. You can log into your pi via SSH or use the app "Cyberduck" to directly download the files from the pi to your pc.

Hope that helped! If you have any other questions, feel free to ask, I'll try to answer.

Rotifox, 2024-11-26:
Perfect, I already have the "klipper.uf2" file downloaded on my PC and my eddy in DFU mode. Now how do I send the downloaded file?

Quinemc, 2024-11-26:
> Hey @Quinemc , I just had the same issue and seem to have fixed it.
>
> Hold down the boot button on your eddy while connecting it to your computer. It should show up as "RP2040" or something similar [...]

Thanks a lot, it works now.

Quinemc, 2024-11-26:
> Perfect, I already have the "klipper.uf2" file downloaded on my PC and my eddy in DFU mode. Now how do I send the downloaded file?

Press boot button and plug in to your laptop/pc usb while holding. The computer will read Eddy as storage then you can kust drag and drop that klipper.uf2 in to the Bootable Eddy. Once pasted, Eddy will restarted itself means successfully flashed into it.

Nevio024, 2024-11-26:
> > Hey @Quinemc , I just had the same issue and seem to have fixed it.
> > Hold down the boot button on your eddy while connecting it to your computer. It should show up as "RP2040" or something similar in your file manager.
> > You can then simply drag over "klipper.uf2". THe file "klipper.uf2" can be found in the folder ~/klipper/out/ on your raspberry pi. You can log into your pi via SSH or use the app "Cyberduck" to directly download the files from the pi to your pc.
> > Hope that helped! If you have any other questions, feel free to ask, I'll try to answer.
> 
> Thanks a lot for your solution, it works now.

Really glad that it works now!

Nevio024, 2024-11-26:
@bigtreetech 
Can you please add a clear guide on how to do this the the instructions on GitHub? 

There have been many people struggling with this, and I think it's important to add it.

Rotifox, 2024-11-26:
Great, it works now. Thanks for sharing the solution. Now follow the configuration steps. It would be ideal if the company that manufactures them added this information to their installation.

Rotifox, 2024-12-01:
Hello. I have the following problem. I hope you can help me. I have this error Error on 'PROBE_EDDY_CURRENT_CALIBRATE': missing CHIP . Any solution?

Hooba97, 2024-12-22:
Eddy Duo same problem
cant get the device ID or serial 
 
I tried to flash the firmware using PC ( flash successful )
 but I cant get it to read by the motherboard 
 eddy is connected to SB2209 USB , how about you?

nitromaroder, 2025-01-07:
@Hooba97 - any luck with your Eddy Duo? I have the same problem - the probe stays in boot mode after reboot / re-flash