bigtreetech/Eddy #97: [Feature request] Redesign board to add an ADXL345

paulstimpson, 2024-11-26:
I request a change to the board design to incorporate an ADXL345 in addition to the levelling sensor. As far as I can tell, no combined levelling and vibration tuning sensor exists. The Eddy is ideally placed to also measure vibration for input shaping. A combined sensor would be a real game changer for me

WayneManion, 2024-11-27:
The Cartographer v3 has an integrated ADXL345.

BTTJUSTIN, 2024-12-05:
Thanks for making this suggestion. We will consider it for future hardware releases. At present, it can be used together with ADXL345 V1.0,

Wrath669, 2024-12-12:
Would not like this as a mandatory include if it adds redundant weight, when some toolheads already have an ADXL345 integrated.

WayneManion, 2024-12-12:
> Would not like this as a mandatory include if it adds redundant weight, when some toolheads already have an ADXL345 integrated.

Not all toolhead boards are mounted in a way that is good for measuring acceleration. The Micro Swiss NG platform springs to mind. An Eddy sensor always has to be mounted parallel to the bed, so that means it would always be a good place for an accelerometer. 

And an ADXL chip weighs like 100mg.