bigtreetech/Eddy #99: BTT Eddy - Failed calibration - incomplete sensor data

effendieffendi, 2024-11-29:
Hi. Today I received a recently purchased Eddy sensor. I was a problem with the calibration. During calibration, after the command "Calibrate_btteddy" (step 5) and aligning the nozzle with the piece of paper, the sensor begins to calibrate and then displays the error "Failed calibration - incomplete sensor data"

Btt Eddy -18.0°C  <<<<<<<<<<<!!!!!!
Btt Eddy Mcu 42.4°C 
Chamber Temp 24.7°C 
Mcu Temp 39.4°C
Nozzle Mcu Temp  38.2°C

Any help???
[image]

effendieffendi, 2024-12-04:
The temperature of BTT Eddy is still below zero. Doesn't this indicate a faulty temperature sensor?
Please watch the attached screen:

[image]

BTTJUSTIN, 2024-12-25:
Thanks for reaching out. Our support team does not operate using the GitHub
platform but we would be happy to help you via email based ticketing system. Please visit
https://biqu.equipment/pages/technical-support and
submit a ticket and we will be in touch with you soon