bigtreetech/Eddy #100: Error on 'PROBE_EDDY_CURRENT_CALIBRATE': missing CHIP

thereealnio, 2024-12-01:
[image]

I have a Voron 2.4r2, with a raspberry pi4 and a btt octopus mainboard and 2x Sovol SV08 printers. Till now, i was using tap on the Voron, and it was working as expected. 
I decided to replace the sensor of one of the SV08 with an eddy usb, and the process was smooth.
I have then, decided to replace the tap on the Voron, and i am getting the following error: Error on 'PROBE_EDDY_CURRENT_CALIBRATE': missing CHIP

I can successfully perform the LDC_CALIBRATE_DRIVE_CURRENT CHIP=btt_eddy .
I get the response: probe_eddy_current btt_eddy: reg_drive_current: 15
The SAVE_CONFIG command will update the printer config file
with the above and restart the printer.

and then i SAVE_CONFIG.

Once the printer restarts, i am sending the PROBE_EDDY_CURRENT_CALIBRATE_AUTO command, the device performs a homing on the X and Y axis, and right after that, it produces the" Error on 'PROBE_EDDY_CURRENT_CALIBRATE': missing CHIP", twice.

Steps i have performed so far:

Tried with a new eddy usb (i had ordered 3), with a different usb cable, and i got the same error message.
Updated everything to the latest version.
I have removed and re-installed, through KIAUH,  Klipper, Moonraker, Mainsail, and Crowsnest.
I have tried different usb ports on the rpi, a different psu (the original from raspberry).
I have tried to re-configure the whole setup from scratch, by re-downloading the latest printer.cfg from BigTreeTech
And, i have tried googling about it, so, rest assured, i will not be getting a response like: "let me google this for you".

At this point, i really can not understand what i am doing wrong, and why i am getting this error... (upgrading everything is not an option, it's no longer "blackfriday" ;-)

Thank you all in advance for your replies

MasterSe7en777, 2024-12-02:
im having the same issue, but the weird thing is, i did get it working once, and then it stopped.... and its back to doing it again...so i have no clue im still trying to figure it out

Rotifox, 2024-12-02:
I'm joining in. I have the same error. I have the USB eddy that seems to be giving a lot of problems.

Darth-Kuni, 2024-12-07:
i get this same error has any one fixed it yet ?

thereealnio, 2024-12-07:
Well... I did...!
After MasterSe7en777 commented and said that it worked once, and since I knew it was not a faulty hardware, I did 2 things. 
1, I replaced the "simple" template, with the "full" one. (Z probe plus experimental settings) 
2, (it might sound stupid.... I made sure, when entering the PROBE_EDDY_CURRENT_CALIBRATE_AUTO
CHIP=btt_eddy
Command, that the second line, is actually in line, and there is no line change there... Like this: PROBE_EDDY_CURRENT_CALIBRATE_AUTO CHIP=btt_eddy

The 2 lines seem identical in the console, but the one on the top is with enter, and the one on the bottom has a simple space.

Since then, it has been working fine for me. I think, we all fell victimes of a bad guide and our copy - paste skills....

IF you also get it working the way I did, please reply, so that people will know!

MasterSe7en777, 2024-12-07:
So I figured out the problem with mine, if you are using canbus you have to remember to remove the 120ohm resistor jumper from your toolboard in my case it was from my sb2209 because the Eddy has a built in 120ohm resistor so it is one end of the can bus and my manta e3ez also has a jumper, you can't have 3 120ohm resistors on the bus...

I don't know what you guys are using but if you have a setup that's the same as mine you need to make sure you remove one of the jumpers and you have to remove the one in the middle of the network which is on the sb2209

thereealnio, 2024-12-07:
I had it on Eddy USB, not the can version

Darth-Kuni, 2024-12-07:
thereealnio thank you so much that worked 

so the PROBE_EDDY_CURRENT_CALIBRATE_AUTO CHIP=btt_eddy has to be on 1 line 
so simple 
eddy duo canbus
m5p

MasterSe7en777, 2024-12-08:
Ok so removing the 120ohm jumper combined with threealnios fix, it's all working perfectly thank you all

closed 2025-01-25 by BTTJUSTIN