bigtreetech/Eddy #104: EDDY DUO on CanBus - Failed calibration - incomplete sensor data

josmir, 2024-12-13:
Dear

I'm unable to move hard with the configuration as I always end-up on Failed calibration - incomplete sensor data every time I run PROBE_EDDY_CURRENT_CALIBRATE_AUTO CHIP=btt_eddy

Can you please help me figure out what's missing or is wrong?
[klippy.log](https://github.com/user-attachments/files/18128700/klippy.log)

Didier-T, 2024-12-15:
Bonjour,
Ta "calibration_temp" est montée plus haut qu'autorisé dans ta section [temperature_probe btt_eddy]

Pour information elle est montée à 31.848116°c

Peut être passer ce paramètre a 33°c et l'ajuster au besoin

josmir, 2024-12-16:
Bonjour Didier

Thank you for having the time to check on this! I've changed the setting as per your sugestion and repeated PROBE_EDDY_CURRENT_CALIBRATE_AUTO CHIP=btt_eddy a few times and ended up on the same result even if the readed temperatures are below 33

Didier-T, 2024-12-16:
Dommage que tu n'ais pas remis le klippy.log

BTTJUSTIN, 2024-12-24:
Thanks for reaching out. Our support team does not operate using the GitHub
platform but we would be happy to help you via email based ticketing system. Please visit
https://biqu.equipment/pages/technical-support and submit a ticket and we will be in touch with you soon