bigtreetech/Eddy #105: Eddy USB calibration issues

Drastil11, 2024-12-17:
I have an Eddy USB, using the latst version of klipper and the eddy probe is flashed with firmware from the latest version of klipper.
Cople of issues that come up.

1.  When I first set the z height and the probe moves to the position to calibrate the sensor, I get a loud screeching noise coming from the extruder stepper and I get a move exceeds maximum extrusion error. I then have to put 200 or so in the maximum extrusion section for the extruder. I no longer get the error but the extruder stepper still makes a god awful sound.
2.  When calibrating the thermal drift compensation, the movement of the probe from the z-height measurement (paper test) position to the probe calibration position is way too slow. By the time the probe itself reaches the position of calibrate for temperature, the temperature has increase 2 or 3 degrees and then, when it moves back for the next temperature interval, it moves so slow that the temperature has climbed beyond the 3 or 4 degree threshold to trigger and z-height measurement. Any way to speed up the movement between paper test and probe calibration?
3.  The following macro (direct from BTT's instructions) always gives me the error "missing chip". I am assuming it has to do with the rawparams portion of the macro. Can I just change that raparams to CHIP=btt_eddy and go from there?
[gcode_macro PROBE_EDDY_CURRENT_CALIBRATE_AUTO]
gcode:
  BED_MESH_CLEAR
  G28 X Y
  G90 # Abs positioning
  G1 X{ printer.toolhead.axis_maximum.x/2 } Y{ printer.toolhead.axis_maximum.y/2 } F6000
  {% if 'z' not in printer.toolhead.homed_axes %}
    SET_KINEMATIC_POSITION Z={ printer.toolhead.axis_maximum.z-1 } # Allows the user to work it down until it touches.
  {% endif %}
  PROBE_EDDY_CURRENT_CALIBRATE {rawparams}

Any help would be greatly appreciated.

dugly66, 2024-12-20:
I got it to work by changing the last line of that macro to this
 PROBE_EDDY_CURRENT_CALIBRATE CHIP=btt_eddy #{rawparams}

Drastil11, 2024-12-21:
Thanks for the reply. Got it working with your suggestion. 
I fixed the other two on my own

closed 2025-01-25 by BTTJUSTIN