bigtreetech/Eddy #107: Eddy Duo ( USB) with the new BTT EBB 2209 USB

Hooba97, 2024-12-22:
following all the steps in both guides for eddy and EBB2209 
I am unable to find the device using `lsusb`  or `ls /dev/serial/by-id/*` 
I flashed the firmware using USB + computer with no change 
do I need to fork to Bigtreetech klipper ?

can you help me?
ps : I am using Manta m8p +CB2

BTTJUSTIN, 2024-12-25:
Thanks for reaching out. Our support team does not operate using the GitHub platform but we would be happy to help you via
email based ticketing system. Please visit
https://biqu.equipment/pages/technical-support and
submit a ticket and we will be in touch with you soon

nitromaroder, 2025-01-07:
I have the same issue with Duo, already opened a case with support - when connecting Duo (set to USB mode of course), it comes up immediately with RP2 Boot enabled (without pressing the button). The flash went successfully, but after flash-reboot it comes back with "RP2 Boot" enabled ("Bus 003 Device 006: ID 2e8a:0003 Raspberry Pi RP2 Boot")

Hooba97, 2025-01-08:
I solved my problem 
I had the wiring wrong when I re crimped the cable

closed 2025-01-29 by Hooba97