bigtreetech/Eddy #108: issues with : PROBE_EDDY_CURRENT_CALIBRATE_AUTO CHIP=btt_eddy

maulionc, 2024-12-23:
Hello 
I dont find the way to calibrate my eddy usb and not sure installation is ok .

what i can check 

Thanks

[image]

BTTJUSTIN, 2024-12-25:
Thanks for reaching out. Our support team does not operate using the GitHub platform but we would be happy to help you via
email based ticketing system. Please visit
https://biqu.equipment/pages/technical-support and
submit a ticket and we will be in touch with you soon

maulionc, 2024-12-30:
Hello Bigtreetech/Eddy

How many times to have answer of my ticket ?

Charles

638Alex638, 2025-01-28:
However, it is written in the screenshot - press the home button on all axes so that the head stands at the specified coordinates.

VDS3D, 2025-04-19:
same issue for me ... do you find a solution ???

maulionc, 2025-04-21:
hello

I try this software and appear very better

https://github.com/3DPrintDemon/Demon_Eddy

why btt don't update somethings?

Charles