bigtreetech/Eddy #109: Rethink how homing is handled

vvuk, 2024-12-26:
I was trying to understand some weirdness with the btt eddy probe. Mainline klipper (from last week), sovol sv08, valerie moore's mount (lower position; the remix'd Eddy Duo mount). BTT Eddy config _without_ the "beta Z offset". I've kept temps consistent at the temp that the eddy was calibrated at; no temp curve in play.

Note that the way homing is handled here is I think broken. Homing is about mapping the printer's stepper position to an actual physical distance. Using the GCODE z_offset to do this is wrong, and then requires gross hacks like the "beta z offset" stuff.

The probe's `z_offset` is set to 2.5, i.e. the probe's trigger point is at 2.5mm off the bed. What I expect to happen: `G28` to home the axes. Then `PROBE_ACCURACY` should report 2.50xxx (i.e. -- when the steppers are at what they think is physical 2.5mm based on the homing, the probe should report 2.5mm).

Instead, what happens is I consistently get numbers from accuracy that are around 2.50-2.65mm, with wide variance. If I use BTT's macros, I can see their macros setting the gcode z offset to things like 0.21, 0.15, 0.05, 0.25, etc.; wildly different value with every Z home. (Can see this by just doing `G28 Z` multiple times.)

`PROBE_ACCURACY` gives very repeatable results (typically stddev 0.001). My theory as to what's happening is that the homing code with the virtual z endstop has some variable latency between the endstop trigger actually happening and it interrupting the homing z axis downward movement.

I've added [reporting the probe accuracy results as status to probe.py](https://github.com/vvuk/klipper/commit/bd34316d5d29d47822150636134ce574f33fbf1b) and I've [changed the homing macro to use it like this](https://github.com/vvuk/sv08-config/commit/bb29fff265f86c34a22d62a63b032d15731a2c01), and I'm getting far better and more accurate results.

bigtreetech, 2024-12-27:
The klipper homing dive only looks for a single trigger and then stops. This is not suitable for an eddy probe that requires an average over several samples. The way that the home therefore works is that it performs a standard home with the knowledge that there will be some error because klipper is only relying on a single sample to trigger. Thereafter it performs a probe sequence in exactly the same xy location which behaves quite differently in that it averages the trigger point over several samples. The probe will be more accurate but it is still looking for a height of 2.5mm to trigger at. Since we know that the homing dive *should* have triggered at 2.5 we can calculate the inaccuracy by subtracting the probe dive from the homing dive results. I.e. if the probe dive triggers at 2.6mm then we know that we were actually 0.1mm high during the home and we need to subtract 0.1mm from the homed z position which is implemented in the SET_Z_FROM_PROBE macro.

Until the entire homing part of klipper can be reworked, this is an effective and transparent solution. Your macro works in a similar manner but it simply performs more probing dives which is not required since the standard probe is already sampling so fast that a single dive already gathers tens of samples and averages them. The only other difference is that your macro is using the SET_KINEMATIC_POSITION gcode which is not recommended relative to the option to simply perform an axis transform as is already being done in the original code.

Z-offsets are a completely different discussion. For more info on how z offsets work and why a standard z-offset no longer applies with an eddy probe see this write up. We are working to update the mainline klipper codebase so that it is better suited to these types of probes without the need for macros.

https://gist.github.com/bigtreetech/484380b26be613b9139bc537510393df

closed 2024-12-27 by bigtreetech

vvuk, 2024-12-27:
> a single dive already gathers tens of samples and averages them

Is this true? The ldc1612 driver uses a single trigger point frequency and does no averaging of the frequency values. [This patch does, though.](https://github.com/KevinOConnor/klipper-dev/commit/beb02f77c8f893da6b2590622294d7bceb758926). The default for probe samples is 1, and ultimately this [goes down into `probing_move` here](https://github.com/Klipper3d/klipper/blob/80d185c94c8dbec7f19d791995242af1bb197057/klippy/extras/probe_eddy_current.py#L350) which does do an average of the frequencies (in `note_probe` -> `_check_samples` -> `_pull_freq`), but it's still only during one dive. Let me know if I'm wrong, though! This code all seems overly complex for what should ultimately be a simple operation, so I'm missing something :)

Also, I don't understand the `+ 0.05` / `+ 0.10` in the start_time/end_time there -- I believe the time units are in seconds? And `trigger_time` is the time the probe actually triggered... why is the code taking the average of the range 50ms-100ms _after_ that to map to the physical toolhead position?

You're right about `PROBE_ACCURACY` though -- I was getting better results with an average, but the right way to do that would be to just increase `samples` under the eddy probe config, no reason to go through the accuracy dance.

> The only other difference is that your macro is using the SET_KINEMATIC_POSITION gcode which is not recommended

Why not recommended? Homing is explicitly the act of setting an accurate kinematic position, so this seems appropriate. It also presents a much better experience to the user -- the z offset value is meaningless if it's set as part of homing, as the user can't think of it as "this is the offset to apply relative to what my hardware thinks is z=0". Instead if they do apply an offset they also need to remember "ok, 0.210 is actually zero" if they want to make adjustments.

And unfortunately the concept of needing to add a z offset still applies; bad thermal calibration/thermal expansion, user preference etc, can all cause someone to want to adjust the z offset on the fly. Sure, the concept of `[probe] z_offset` doesn't apply (it should really be named `probe_trigger_z` or something), but

vvuk, 2025-01-13:
@bigtreetech just FYI -- I've mostly rewritten the eddy current sensor code & some of the LDC code in order to clean up the code and implement "tap" functionality for precise z-offset. There's a [readme document here](https://github.com/vvuk/klipper/blob/vlad/eddy-ng/README_eddy_ng.md) and that branch has the code. I'm in the process of gathering more testers and data, though it's working well for me with a BTT Eddy Duo (but the coil design I think is somewhat limiting). Feel free to reach out if you're interested in discussing more, either on discord (@vladv) or email (vladimir at pobox dot com).