bigtreetech/Eddy #112: A lot of Jumps in scan

klich3, 2025-01-02:
Hi team, when I run the calibration of the table with Eddy I get lines with jumps in each line that scans.
The Eddy is at a height of 2cm from the table adjusted as in the manual, with the 3D printed model.

Maybe there is a problem with Fimware?

[image]

klich3, 2025-01-02:
For more information it is an Eddy V1 coil I have it connected by USB.

```text
[probe_eddy_current btt_eddy]
sensor_type: ldc1612
i2c_mcu: eddy  # This value is good for the Eddy USB but would need to be adjusted for the Eddy Coil according to the MCU you have used.
i2c_bus: i2c0f # This value is good for the Eddy USB but would need to be adjusted for the Eddy Coil according to the I2C bus you have used.
# Measure the offsets below using the method described here: https://www.klipper3d.org/Probe_Calibrate.html#calibrating-probe-x-and-y-offsets
# For a standard Voron stealthburner X carriage mount there should be no need to change the defaults below.
x_offset: -27
y_offset: 0

# This section is only relevant for Eddy USB. Comment it out for Eddy Coil.
[temperature_probe btt_eddy]
sensor_type: Generic 3950
sensor_pin: eddy:gpio26
horizontal_move_z: 2
```

klipper branch v0.12.0-322-g0855994e
eddy rp2040 Version: v0.12.0-166-g957475421

grahamwilliams, 2025-01-03:
The eddy's at 20mm for the calibration, but mounted ~2mm-5mm above the nozzle, right?

klich3, 2025-01-05:
> The eddy's at 20mm for the calibration, but mounted ~2mm-5mm above the nozzle, right?
Correct, in my case ~5mm 

Could it be because I used the magnetic sticker for the table?

klich3, 2025-01-16:
Thank you Graham, This afternoon I will adjust and calibrate it again as you told me.

klich3, 2025-01-21:
Remains unchanged

klich3, 2025-02-08:
Now I update to latest version of klipper and same for Eddy
Version: v0.12.0-418-g0114d72a6

The fast scanning is done by points, not line by line as before. 

[image]

bigtreetech, 2025-02-08:
It is likely that you are not using the correct bed mesh scanning type. We recommend using the new firmware which is available with instructions here:

https://github.com/vvuk/eddy-ng/tree/main

klich3, 2025-02-10:
Thanks for response, tonight I implement it and test it.

klich3, 2025-02-12:
Now I install eggy-ng, has a totally different method of calibration, but in the end I was able to do it without problems, I still have to investigate all the commands. I started the scan with “rapid scan” correctly. The jumps have decreased but are still there.

Regards

klich3, 2025-03-06:
The curious thing is that in reality the lines do not make much sense, I attach an image of where the scan starts and where it ends.

[image]

1st is first point or screw base screw, 4th is final. Blue line are the head passes the route...

Hooba97, 2025-03-07:
i had a similar problem when i first installed EDDY, 
it tuned out to be a motion system problem and the toolhead not secure 
with eddy-ng i get perfect first layers 
i hope that helps 

[image]

dansdevelops, 2025-03-07:
> I attach an image of where the scan starts and where it ends.
I have doubts about that. Klipper always scans along X axis then steps to next Y coordinate and changes move direction. I even looked at the point generation code.