bigtreetech/Eddy #115: Unable to exit DFU RP2 boot mode on Eddy Duo

nitromaroder, 2025-01-08:
Hi all,

I am stuck in the boot mode while trying to flash Klipper FW:
- flash is done via the delivered USB cable
- as soon as the Eddy Duo gets connected, it is immediately in RP2 boot mode ("2e8a:0003 Raspberry Pi RP2 Boot")
- compiling and flashing Klipper FW reboots the probe, but it comes up back in DFU mode again

Klipper FW:
[image]

Flashing Klipper FW:
```
make flash FLASH_DEVICE=2e8a:0003
  Flashing out/klipper.bin to 2e8a:0003
sudo lib/rp2040_flash/rp2040_flash out/klipper.bin

Loaded UF2 image with 0 pages
Found rp2040 device on USB bus 3 address 5
Flashing...
Resetting interface
Locking
Exiting XIP mode
Erasing
Flashing
Rebooting device
```

Reboot and re-connect in DFU mode after successful flash:
```
[ 1060.830767] usb 3-1.1: USB disconnect, device number 5
[ 1061.672740] usb 3-1.1: new full-speed USB device number 6 using ehci-platform
[ 1061.894263] usb 3-1.1: New USB device found, idVendor=2e8a, idProduct=0003, bcdDevice= 1.00
[ 1061.894291] usb 3-1.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 1061.894297] usb 3-1.1: Product: RP2 Boot
[ 1061.894302] usb 3-1.1: Manufacturer: Raspberry Pi
[ 1061.894308] usb 3-1.1: SerialNumber: E0C9125B0D9B
[ 1061.895094] usb-storage 3-1.1:1.0: USB Mass Storage device detected
[ 1061.895840] scsi host0: usb-storage 3-1.1:1.0
[ 1062.909696] scsi 0:0:0:0: Direct-Access     RPI      RP2              3    PQ: 0 ANSI: 2
[ 1062.910153] sd 0:0:0:0: Attached scsi generic sg0 type 0
[ 1062.912200] sd 0:0:0:0: [sda] 262144 512-byte logical blocks: (134 MB/128 MiB)
[ 1062.913636] sd 0:0:0:0: [sda] Write Protect is off
[ 1062.913664] sd 0:0:0:0: [sda] Mode Sense: 03 00 00 00
[ 1062.914981] sd 0:0:0:0: [sda] No Caching mode page found
[ 1062.915006] sd 0:0:0:0: [sda] Assuming drive cache: write through
[ 1062.930088]  sda: sda1
[ 1062.934199] sd 0:0:0:0: [sda] Attached SCSI removable disk
```

I tried without success also to flash "klipper.uf2" via copying the file to the Eddy Duo storage with same results. The DFU button is not pressed.

MoustacheMonkey, 2025-07-25:
I have the exact same problem. The Eddy Duo keeps starting in boot mode, this started after powering up the printer, the Eddy probe worked fine in the past days however.. 

Behaviour is the same as in your description. It gets attached as a storage device. Klipper can't connect because it is not listed in the serial devices directory.

I can see the device between the USB devices:
`Bus 003 Device 013: ID 2e8a:0003 Raspberry Pi RP2 Boot`

it is not listed between the serial devices :
`biqu@SV08:~/klipper$ ls /dev/serial/by-path/*
/dev/serial/by-path/platform-5101400.usb-usb-0:1:1.0
/dev/serial/by-path/platform-5200400.usb-usb-0:1:1.0
`

bigtreetech, 2025-08-04:
This could be due to two things:

1. A recent change in klipper that sets the flash frequency higher than the flash chip can handle. To fix it you need to compile the firmware using CLKDIV 4 instead of CLKDIV 2 in the menuconfig.
2. Check that the button at the top of the Eddy is not stuck.

Here is a pic of the WRONG clkdiv within menuconfig:

[image]

khisanthax, 2025-08-04:
I used clkdiv 4 and I'll check to make sure the button is pressed.  I'm in
klipper .13, so I'll try to update and reflash.  Is it possible that a high
drive current can damage it?

bigtreetech, 2025-08-05:
Ah wait. Are you using a bootloader offset? If so, turn it off and reprogram.

khisanthax, 2025-08-06:
Is there anyway to send a command to exit boot mode?

khisanthax, 2025-08-06:
I was plugging in the connector to the Eddy after it had already been connected to the PC first.  Could I have done that and caused a short by inserting the connector at an angle and shorting some pins?

sinhuellas86, 2025-08-08:
hello i have same problem
At first I used an eddy, and the printer worked perfectly, but one day, while I was homing, I turned off the printer because I had to go. When I restarted it, the printer stopped recognizing the eddy. Through PuTTY, it appears to me in DFU mode like yours. I bought another one, tried compiling it, and it still wouldn't exit DFU mode... In the end, I have two eddys that don't work.

I have gone back to using the one that came from the factory, which at least doesn't cause any problems, but if anyone knows how to exit boot mode, I would appreciate it.

bigtreetech, 2025-08-08:
> hello i have same problem At first I used an eddy, and the printer worked perfectly, but one day, while I was homing, I turned off the printer because I had to go. When I restarted it, the printer [...]

Have you tried the recommendations made in this thread about CLKDIV and the bootloader offset?

khisanthax, 2025-08-08:
When plugging in the cable, which side did you plug in first?

sinhuellas86, 2025-08-09:
> Have you tried the recommendations made in this thread about CLKDIV and the bootloader offset?

Yes, I installed version 4, but when I installed it, it restarts but remains in boot mode. I tried installing it from the PC and it restarts but returns to boot mode without any change.

costads, 2025-08-09:
Mine started to do that as well after a reboot. Constantly in DFU mode.
Tried to reflash with both V2 and V4 without luck :/

bigtreetech, 2025-08-09:
> Yes, I installed version 4, but when I installed it, it restarts but remains in boot mode. I tried installing it from the PC and it restarts but returns to boot mode without any change.

What about the bootloader offset? Did you also try that?

costads, 2025-08-09:
I followed a guide when I flashed it, and he was using CLKDIV 2. So my problem began before even trying CLKDIV 4.
The first message I got was that Klipper lost access to the MCU, and then after the reboot, it's staying in DFU mode.

sinhuellas86, 2025-08-09:
> What about the bootloader offset? Did you also try that?

yes and nothing i put no bootloader