bigtreetech/Eddy #117: Temperature Compensation questions

SRAsolution, 2025-02-05:
Hi everyone,

I'm currently using Eddy Coil and I'm interested about Temperature Compensation. 

Questions:
-why eddy coil cannot perform Temperature Compensation?
- what happens if I don't have Temperature Compensation?... the nozzle will be too far or to close if the set tempereature is over the calibrated tempereature. 
-can the Temperature Compensation be performed manually with a macro? 

Thanks for the support

looxonline, 2025-02-05:
It does not have a thermistor. There is no capacity on the wiring harness for the eddy coil to carry the thermistor signal. Soon tap functionality for the Eddy will be released and then thermal compensation will not matter anymore. You can test the beta tap functionality yourself here: https://hackmd.io/yEF4CEntSHiFTj230CdD0Q?view#26-Jan-2025

SRAsolution, 2025-02-05:
Thanks you for the quick explanation and for the link.
I forgot to mentioned that the Eddy Coil is connected to EBB42 which currently have the pt100/1000 slot still available so the question is can we use this thermistor to support Eddy Coil calculation?

looxonline, 2025-02-05:
No. That will not work. The thermistor needs to be directly by the coil.

closed 2025-02-05 by looxonline

SRAsolution, 2025-02-05:
Okay, now I understand the basics. I will test the beta tap ASAP and thank you