bigtreetech/Eddy #118: Variables is not valid config Z offset

KD-4-Lyfe, 2025-02-06:
Been having issues with the z offset. Only way to get it good is quite of a tight pinch on the paper. If I baby stepping and save offset I get the error:
Section 'variables' is not a valid config

If I go into variables.cfg and delete all saved in there it works again.

I can baby step and get a perfect offset before each print but can't save anything.

Using eddy duo on CAN with ebb36

[image]

KD-4-Lyfe, 2025-02-06:
line 1342

> Config error
Traceback (most recent call last):
  File "/home/biqu/klipper/klippy/klippy.py", line 130, in _connect
    self._read_config()
  File "/home/biqu/klipper/klippy/klippy.py", line 127, in _read_config
    pconfig.check_unused_options(config)
  File "/home/biqu/klipper/klippy/configfile.py", line 497, in check_unused_options
    self.validate.check_unused(config.fileconfig)
  File "/home/biqu/klipper/klippy/configfile.py", line 441, in check_unused
    raise error("Section '%s' is not a valid config section"
configparser.Error: Section 'variables' is not a valid config section
webhooks client 3917410416: New connection
webhooks client 3917410416: Client info {'program': 'Moonraker', 'version': 'v0.9.3-29-g549f9fe'}
Unable to issue reset command on MCU 'mcu'
Unable to issue reset command on MCU 'eddy'
Unable to issue reset command on MCU 'EBBCan'
webhooks client 3917410416: Disconnected
Restarting printer
Start printer at Thu Feb  6 07:22:56 2025 (1738826576.5 4905.5)

[klippy (25).log](https://github.com/user-attachments/files/18688362/klippy.25.log)

looxonline, 2025-02-06:
Please follow this guide closely. It will give you access to the public beta for the tap functionality which removes the need to adjust z-offset. https://github.com/vvuk/eddy-ng/wiki

wysockiw, 2025-02-23:
I have the same problem, installed latest version of eddy and latest official document says exactly that z-offset setting is supported ! 
@looxonline Please follow this guide closely -> https://github.com/bigtreetech/Eddy?tab=readme-ov-file#z-offset
When you save config then you klipper is throwing an errror as above .

bigtreetech, 2025-02-24:
> I have the same problem, installed latest version of eddy and latest official document says exactly that z-offset setting is supported ! [@looxonline](https://github.com/looxonline) Please follow [...]

Did you follow the guide that he pointed to? Doing so will show you how to use tap functionality and reintroduce z-offset.

wysockiw, 2025-02-24:
Yeah, but I thought this section was meant to address issues with the Eddy probe, not to point to other solution scripts or separate repositories / solutions.

The steps you provided for Eddy simply don’t work— it cause the variables.cfg file to be overwritten and break Klipper. Therefore, these steps should either be fixed or removed from the official documentation, as they currently do not work. The link provided by @looxonline is excellent, but it takes a completely different approach that requires both TAP and Eddy.

bigtreetech, 2025-02-24:
> Yeah, but I thought this section was meant to address issues with the Eddy probe, not to point to other solution scripts or separate repositories / solutions.
>
> The steps you provided for Eddy simply [...]

That link reflects the new tap method for Eddy. It is being incorporated into the official documentation due for release late next week.