bigtreetech/Eddy #119: intermittent "Error during homing z: Communication timeout during homing"

Joshua1859, 2025-02-06:
When homing I'm getting (sometimes) this error but i simply home the printer or restart the print and seems to work, what's the problem? I don't understand.

bigtreetech, 2025-02-06:
Likely not related to the probe. This seems like a comms issue with your setup. Lots of info here. https://klipper.discourse.group/t/canbus-communication-timeout-while-homing-z/3741/60

closed 2025-02-06 by bigtreetech

Joshua1859, 2025-02-06:
I'm using usb with eddy duo.

bigtreetech, 2025-02-06:
Have you checked your logs to look for lost packets? Not only that but your SBC may be overloaded and causing packets to wait in the return queue for too long. What SBC are you using?

Joshua1859, 2025-02-06:
I'm using a PI 2. I can't seem to find anything in the log but im not quite sure what to look for.

Joshua1859, 2025-02-06:
And what i find strange is that most of the time on the second maybe third time at the worst it seems to work. I think the further away from the bed the more times I have to do the homing again.

bigtreetech, 2025-02-06:
Yea, that is something that I have personally had in the past too. It can be very difficult to track down but it is generally related to messages waiting in the incoming queue for too long.

Is it a BTT Pi2? If so then we have a new image that is currently in beta. We will be making releases rapidly over the next two weeks until we get to the final version. The linux kernel is far more efficient and maybe it will help with this issue.

Joshua1859, 2025-02-06:
No a Raspberry Pi 2.

bigtreetech, 2025-02-06:
> No a Raspberry Pi 2.

Oh OK. In that case the issue is almost certainly your pi. It is just too slow to handle the incoming messages.

Joshua1859, 2025-02-06:
Oh ok I'll have to use a phyical endstop instead.

bigtreetech, 2025-02-06:
One thing that you may want to try is to use the lite version of raspbian. If you are using the desktop version then there will be a lot of resources that are eaten needlessly. 

[image]

motogon, 2025-04-05:
I have the same problem , I regularly get an error in klipper "!!!! Error during homing z: Communication timeout during homing" - and because of this , I can not use the printer with your device . Connection via usb . 

I have klipper running on Host (x86_64, 64bit)
Version: v0.12.0-458-gd886c1761
OS: Debian GNU/Linux 12 (bookworm)
(virtual machine)

Tried it on a single hardware computer where klipper is the only software. 

In both cases - the problem with your device is repeated. 
In several attempts - 1 out of 10, it works properly, in other cases - an error.