bigtreetech/Eddy #124: Z offset always off

nipunarora8, 2025-03-07:
I am using eddy probe for z endstop, homing and beta z offset but it is always off, always off. Even without using eddy probe for z offset it is still always off. I don't understand why.

awiltschek, 2025-03-11:
Whats missing in the docu is that after babystepping the nozzle to the bed - YOU DONT PRESS THE GUIS SAVE BUTTON !
Instead you need to manually enter the commands

Z_OFFSET_APPLY_PROBE
SAVE_CONFIG

This article solved it for me but not the docu :-) this is wrong
https://klipper.discourse.group/t/klipper-not-applying-z-offset-to-future-prints/13317/4

nipunarora8, 2025-03-11:
Hi @awiltschek thanks for your reply. 
But idk how but the beta z-offset seems to work now. All I did was z offset calibration and after that it is working. The first layer is rather little too crisp but I think it should do the job. I will still wait for a week and many different prints to confirm it. 
Thanks

Brutalant, 2025-04-14:
> The docu is incorrect ! instead of pressing the save button on the GUI you need to send Macro "Z_OFFSET_APPLY_PROBE" to store the babystepped value into the variables.cfg file so that the G28 can read any use it ! Thats why i made a SAVE_Z_OFFSET macro to use it on the GUI instead of the Save Button at the Toolhead.

Had a feeling the klipper save Z wasn't working thanks for this info will try tomorrow

nipunarora8, 2025-04-14:
I stopped using eddy probe, took it off and put back my biqu micro probe. 
Life has been pretty smooth since then.

wsturner88, 2025-05-11:
> The docu is incorrect ! instead of pressing the save button on the GUI you need to send Macro "Z_OFFSET_APPLY_PROBE" to store the babystepped value into the variables.cfg file so that the G28 can [...]

Would you mind sharing your Macro?   I don't have the Z_OFFSET_APPLY_PROBE in my GUI.  I guess I could just type it.

awiltschek, 2025-05-11:
you have it in the default macros of bigtreech. you make a small macro which is calling the bigtreetech macro for Z_OFFSET_APPLY_PROBE and add it you your gui. thats it.

Nuttavoot, 2025-05-25:
I found that "save" button near Z-offset settings already does 2 commands in sequence "Z_OFFSET_APPLY_PROBE" then SAVE_CONFIG already.

Bodya111, 2025-12-27:
And so I also encountered this problem, here is my solution after you tried printing a square in 1 layer by manually calibrating the height to reach the one that seems good, remember how much -0.??? And then carry out a new calibration The first is to heat the table and nozzle to the printing temperature, the second is G28 (homing) then do the command PROBE_EDDY_CURRENT_CALIBRATE CHIP=btt_eddy, And now bring it to 0.000 and subtract the value you remembered, now just let the printer calibrate and you're done!