bigtreetech/Eddy #126: klipper Version: v0.12.0-458-gd886c1761

Val-dose-things, 2025-03-19:
klipper Version: v0.12.0-458-gd886c1761 is not compatible with eddy. just spent 3 hours flashing and re flashing eddy and after restarting my printer it could not find the probe. 

to down grand run 

 cd ~/klipper
sudo systemctl stop klipper
git reset --hard HEAD~15 

then 

make clean
make menuconfig

make

sudo service klipper stop
make flash FLASH_DEVICE=2e8a:0003
sudo service klipper start

should fix our issue if you have it also.

Val-dose-things, 2025-03-19:
```
BTT EDDY USB

Version: v0.12.0-458-gd886c176

starts with no issue after flashing
power off and on 

Klipper reports: ERROR

mcu 'eddy': Unable to connect
Once the underlying issue is corrected, use the
"FIRMWARE_RESTART" command to reset the firmware, reload the
config, and restart the host software.
Error configuring printer
-------------------------------------------------------------------
Version: v0.12.0-440-g53f1bf2a

starts with no issue after flashing
power off and on 

starts up with no problems
```
fix was
cd ~/klipper
sudo systemctl stop klipper
git reset --hard HEAD~18

nefelim4ag, 2025-03-22:
I suspect this is caused by: https://github.com/Klipper3d/klipper/pull/6829

Val-dose-things, 2025-03-22:
Yeah seems like it. Need to see if the new version fixed it.

nefelim4ag, 2025-03-22:
> Need to see if the new version fixed it.

I may suggest taking a look at the discourse link above, I can suggest only 2 things to look at/test.

Val-dose-things, 2025-03-22:
Ok so 1 v0.12.0-441-gedc3d34be  seems to be the last working version I'll test the new version soon.

recliq, 2025-03-23:
Try compiling with `CLOCKDIV 4`, that finally did the trick for me.
thx @nefelim4ag 
```
[*] Enable extra low-level configuration options
    Micro-controller Architecture (Raspberry Pi RP2040/RP235x)  --->
    Processor model (rp2040)  --->
    Bootloader offset (No bootloader)  --->
    Flash chip (GENERIC_03H with CLKDIV 4)  --->
    Communication Interface (USBSERIAL)  --->
    USB ids  --->
[*] Optimize stepper code for 'step on both edges'
()  GPIO pins to set at micro-controller startup
```

B2Hansen, 2025-05-03:
Thanks. lots of time wasted until i found this ! changed to eddyng and i thought it was the problem.