bigtreetech/Eddy #131: Eddy Duo wont work in a preheated printer, just throws {Error during homing z: Eddy current sensor error}

PigeonFX, 2025-04-22:
OK, having a nightmare with this probe. 

It's mounted on the stock Voron 2.4 X carriage, and the probe is 2.5mm away from the bed with the nozzle touching the bed. 

I have installed the probe, run the command `LDC_CALIBRATE_DRIVE_CURRENT CHIP=btt_eddy` reg_drive_current = 20 (it always reads 20 regardless of where it is in relation to the bed, not sure if that is an issue, but thought it was worth mentioning as most post talk about 15 & 16)

Then I run the `PROBE_EDDY_CURRENT_CALIBRATE_AUTO CHIP=btt_eddy` and `TEMPERATURE_PROBE_CALIBRATE PROBE=btt_eddy TARGET=65 STEP=5` (as the probe can reach 60c before the print starts considering its sitting 10mm near the bed).

At room temperature, the probe can home and QGL without any issue, but as soon as I heat the bed and the probe gets over 40c it will fail to home, throwing a `Error during homing z: Eddy current sensor error`. 

Here are all the auto-generated results from Printer.cfg:

> #*# [probe_eddy_current btt_eddy]
> #*# reg_drive_current = 20
> #*# calibrate =
> #*# 	0.050000:3508629.971,0.090000:3508081.455,0.130000:3507531.524,
> #*# 	0.170000:3507050.147,0.210000:3506480.323,0.250000:3505997.291,
> #*# 	0.290000:3505513.600,0.330000:3504996.427,0.370000:3504508.424,
> #*# 	0.410000:3504011.938,0.450000:3503590.258,0.490000:3503099.728,
> #*# 	0.530000:3502658.069,0.570000:3502211.575,0.610000:3501775.165,
> #*# 	0.650000:3501361.915,0.690000:3500901.292,0.730000:3500503.795,
> #*# 	0.770000:3500102.105,0.810000:3499680.692,0.850000:3499291.612,
> #*# 	0.890000:3498936.789,0.930000:3498548.189,0.970000:3498135.379,
> #*# 	1.010000:3497787.189,1.050000:3497455.085,1.090000:3497066.554,
> #*# 	1.130000:3496725.939,1.170000:3496362.127,1.210000:3496039.446,
> #*# 	1.250000:3495691.730,1.290000:3495375.967,1.330000:3495052.856,
> #*# 	1.370000:3494741.013,1.410000:3494426.521,1.450000:3494135.976,
> #*# 	1.490000:3493820.817,1.530000:3493522.242,1.570000:3493241.569,
> #*# 	1.610000:3492944.455,1.650000:3492668.364,1.690000:3492381.313,
> #*# 	1.730000:3492118.859,1.770000:3491854.826,1.810000:3491591.227,
> #*# 	1.850000:3491330.039,1.890000:3491075.687,1.930000:3490811.423,
> #*# 	1.970000:3490597.446,2.010000:3490322.031,2.050000:3490063.205,
> #*# 	2.090000:3489851.960,2.130000:3489625.451,2.170000:3489413.172,
> #*# 	2.210000:3489198.378,2.250000:3488974.167,2.290000:3488724.180,
> #*# 	2.330000:3488504.783,2.370000:3488302.306,2.410000:3488109.711,
> #*# 	2.450000:3487902.094,2.490000:3487707.549,2.530000:3487532.006,
> #*# 	2.570000:3487325.299,2.610000:3487116.274,2.650000:3486933.004,
> #*# 	2.690000:3486762.440,2.730000:3486582.048,2.770000:3486370.148,
> #*# 	2.810000:3486193.237,2.850000:3486071.069,2.890000:3485879.980,
> #*# 	2.930000:3485691.451,2.970000:3485560.085,3.010000:3485379.052,
> #*# 	3.050000:3485249.409,3.090000:3485047.857,3.130000:3484915.026,
> #*# 	3.170000:3484752.099,3.210000:3484605.084,3.250000:3484464.140,
> #*# 	3.290000:3484301.485,3.330000:3484164.804,3.370000:3484016.977,
> #*# 	3.410000:3483870.467,3.450000:3483755.881,3.490000:3483624.417,
> #*# 	3.530000:3483490.661,3.570000:3483338.079,3.610000:3483199.062,
> #*# 	3.650000:3483076.698,3.690000:3482946.142,3.730000:3482814.198,
> #*# 	3.770000:3482685.877,3.810000:3482579.263,3.850000:3482433.141,
> #*# 	3.890000:3482320.905,3.930000:3482214.584,3.970000:3482102.367,
> #*# 	4.010000:3481970.651,4.050000:3481867.708
> #*#
> #*# [temperature_probe btt_eddy]
> #*# calibration_temp = 40.302138
> #*# drift_calibration =
> #*# 	3560420.891725, -1796.392185, 11.230520
> #*# 	3552392.134836, -1715.440351, 10.553839
> #*# 	3545939.427858, -1652.541517, 10.041377
> #*# 	3540239.034980, -1583.751777, 9.476795
> #*# 	3535481.266169, -1520.437856, 8.929055
> #*# 	3531903.798937, -1481.600144, 8.609010
> #*# 	3529270.477943, -1463.036945, 8.477150
> #*# 	3526710.580725, -1433.378008, 8.247333
> #*# 	3524861.062135, -1418.396849, 8.129483
> #*# drift_calibration_min_temp = 37.93912216136242
>

bigtreetech, 2025-04-22:
20 seems a bit high. You could try to raise the probe just a little more or manually reduce the LDC drive current to the 15-16 range to see what happens. There is also a new firmware plugin that will offer you tap functionality and help you with this calibration here: https://github.com/vvuk/eddy-ng

bigtreetech, 2025-04-22:
> As I mentioned in my first post, raising or lowering the Eddy doesn't change the LDC drive current at all; it is 20 when touching the bed and 20 when Z is at 250mm. It is always 20.
> 
> If I manually change it to 16 in the printer.cfg, I get `Error during homing z: Eddy current sensor error` even if I run `PROBE_EDDY_CURRENT_CALIBRATE_AUTO CHIP=btt_eddy` after setting LDC drive current manually to 16.

To be 100% sure, your mount is completely stock with no metal bits and pieces around it?

PigeonFX, 2025-04-22:
> To be 100% sure, your mount is completely stock with no metal bits and pieces around it?

Yes, totally stock, no metal round it apart from the E3D Voron Revo hotend and the 2 mounting scrows that came with it, Eddy Duo connected to the EBB 2240 2209 all configured for CAN with Katapult.

bigtreetech, 2025-04-22:
I feel like there is something not right with that Eddy if it is asking for an LDC drive current of 20 regardless of installation height. I would log a return. Feel free to point to this thread in the return.

PigeonFX, 2025-04-22:
OK, will grab another one from Amazon for tomorrow and arrange a return of this one to Biqu. 

Thank you for your help. I will report back when the replacement is fitted, and hopefully we can close this ticket.

bigtreetech, 2025-04-22:
Cool. Use the ticket system on the BIQU site to log the return. Remember that they are probably using a translator so try to use bullet points to explain what you've done so far and then also point them to this thread. Include pics of the calibration spitting back 20 and the error that it throws. It all helps the team to see clearly that you've read the manual and followed the direction but have had an issue.