bigtreetech/Eddy #133: Eddy Duo CAN - Install Questions

wsturner88, 2025-05-04:
Hoping for some help in getting the Eddy Duo Can connected to the CAN on the SB2209 Daughter Board.

1) Flashed Katapult to the Eddy Can (Switch in USB Mode)
2) Installed the Eddy Duo to the Daughter Board of the SB2209 (Switch on Duo is now CAN)

DO I REMOVE THE 120R resistor from the SB2209???

Via terminal:
python3 ~/katapult/scripts/flashtool.py -i can0 -q

I get no Nodes.. 

Any recommendations?

bigtreetech, 2025-05-04:
What comms did you configure katapult for in the make menuconfig?

wsturner88, 2025-05-04:
[image]

bigtreetech, 2025-05-04:
> [image]

The status LED should be blinking once per second if it has entered into katapult. You also need to ensure that the baud rate in the can 0 config file on your SBC is set to the same baud rate as the Eddy (1Mbps). If the status LED is not blinking then katapult may think that there is a valid application in the memory and be jumping into it. You would need to erase the whole memory of the RP2040 to prevent that and I would recommend using the USB interface for that purpose.

wsturner88, 2025-05-04:
Thank you for the quick reply.. 
Light on Eddy Duo CAN is solid

When I flash Katapult, do I have the switch on CAN or USB?   I originally flashed Katapult via the USB on the Rpi (Switch USB) then changed to CAN and installed onto the CAN bus.  Correct?

How to you ensure you erase the memory on the RP2040?

bigtreetech, 2025-05-04:
> Thank you for the quick reply.. Light on Eddy Duo CAN is solid
>
> When I flash Katapult, do I have the switch on CAN or USB? I originally flashed Katapult via the USB on the Rpi (Switch USB) then [...]

Flashing anything onto the Eddy should be done via USB unless you are using katapult and have successfully managed to get it to work previously. Until then, all should be via USB.

Process would be to flash via USB but configure the katapult bootloader to communicate via CAN in the future (as you have done). Also set the CLKDIV to 4 instead of 2. There was a recent change in klipper that broke compatibility with some flash chips and now this extra setting is required. We will be updating our documentation accordingly.

[image]

To erase the chip memory follow the instructions here: https://www.raspberrypi.com/documentation/microcontrollers/pico-series.html#resetting-flash-memory

wsturner88, 2025-05-04:
THANK YOU!   The CLKDIV was the issue.. Below is the Katapult config that worked

ANOTHER NOTE - ON YOUR SB2209 LEAVE R120 Jumper on.

[image]

closed 2025-05-04 by bigtreetech

bsafh, 2025-12-02:
sorry to re-open this one, but

in https://canbus.esoterical.online/toolhead_flashing/common_hardware/BigTreeTech%20Eddy%20Duo/README.html which usually a good source they say for katapult 

```
bootlaoder = 16 KiB
and CLKDIV 2
```
is that outdated?

[image]

for klipper

[image]