bigtreetech/Eddy #143: Tap mount useless?

jangrewe, 2025-08-22:
I just rebuilt my toolhead to use a ChaoticLab Tap V2 and also bought an Eddy Duo as i saw that there is an "official" mount in this repo: https://github.com/bigtreetech/Eddy/tree/master/3D/EDDY_PRINT_PARTS/Eddy_TapV2

So i printed that, bolted it onto the Tap V2, installed the Tap V2 to my X axis, and then noticed that the Tap V2 is now not a Tap anymore, but just as toolhead bracket, as it doesn't move up anymore - and that's even without the cables installed!

So i'm wondering what the point of the mount for the Tap V2 is, if it doesn't tap anymore after adding Eddy to it?

jangrewe, 2025-08-22:
How is the firmware for the probe related to the mount you're providing, and that it physically(!) stops the toolhead from being able to move up and down on the Tap rail? 

One is software, the other is hardware....

bigtreetech, 2025-08-22:
Personally, I can't speak for your issues with the mount as I have never used it in an individual capacity. Github is here primarily to monitor firmware issues which is what I am more familiar with (not that I don't want to help more but I am just explaining the process).

Eddy-ng is firmware that turns the eddy into a tap probe with incredible accuracy. It completely negates the need for the CNC Tap V2. Nevertheless, I am 100% confident that the mount has been tested as I recall a trusted colleague designing and testing it. If you don't like the idea of eddy-ng then I would recommend contacting the official support channel using the ticketing system on our website where someone will be better geared to assist. Alternately, you could also consider this as you may just be using a hotend that is incompatible with the BTT mount (also completely untested by me in an individual capacity): https://www.printables.com/model/1011227-btt-eddy-mount-for-chaoticlab-tap-v2

closed 2025-08-22 by bigtreetech